mainsail-crew/mainsail #124: Unable to delete STL files

nife87, 2020-11-10:
**Describe the bug:**\
STL files (accidentally) uploaded via Mainsail v0.2.6 are not removable (causes a delete error). They are still removable via CLI, of course.
Solution: Optional extension filter in upload dialog and/or ability to delete STL/any files.

**Expected behavior:**\
- Upload an STL file via Mainsail.
- Right click and press Delete.
- Error message pops up in the bottom-right corner.

**Screenshots:**\
[image]

**Logfiles:**\
[moonraker.log](https://github.com/meteyou/mainsail/files/5517937/moonraker.log)
[klippy.log](https://github.com/meteyou/mainsail/files/5517938/klippy.log)

meteyou, 2020-11-13:
thx for the bugreport... looks like a charset problem with the [] chars.

```
2020-11-10 14:28:28,104 [web.py:log_request()] - 500 DELETE /server/files/gcodes/[a]_80t_hubbed_gear_x4_rev1.stl (192.168.1.51) 188.24ms
2020-11-10 14:28:54,195 [web.py:log_exception()] - Uncaught exception DELETE /server/files/gcodes/[a]_80t_hubbed_gear_x4_rev1.stl (192.168.1.51)
HTTPServerRequest(protocol='http', host='192.168.1.62', method='DELETE', uri='/server/files/gcodes/[a]_80t_hubbed_gear_x4_rev1.stl', version='HTTP/1.0', remote_ip='192.168.1.51')
```

nife87, 2020-11-13:
If you are referring to the [] in the filename, then no, they are not the issue. I have the same issue with all filenames (tried quite a few), for instance CubePuzzle.stl.
[moonraker.log](https://github.com/meteyou/mainsail/files/5538956/moonraker.log)

The corresponding parts:
`2020-11-13 19:28:18,360 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"server.files.metadata","params":{"filename":"CubePuzzle.stl"},"id":5627}
2020-11-13 19:28:18,361 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": 404, "message": "Metadata not available for <CubePuzzle.stl>"}, "id": 5627}
2020-11-13 19:28:23,840 [web.py:log_exception()] - Uncaught exception DELETE /server/files/gcodes/CubePuzzle.stl (192.168.1.51)`

meteyou, 2020-12-10:
i cannot reproduce this problem with a current installation.

closed 2020-12-10 by meteyou