mainsail-crew/mainsail #126: [FR] Add mouseover text describing function of GUI icons

yonkiman, 2020-11-12:
**Is your feature request related to a problem? Please describe.**
Upgraded from 0.2.4 to 0.2.6.  Was looking for the slider to speed up the current print and couldn't initially find it (don't remember exactly what it looked like in 0.2.4, but it was easy to find).  I believe that the speed slider now has a clock icon, the extrusion multiplier has squiggly lines, etc., but there's no universal standard for these icons, so I was a little uncomfortable initially using them in case I'd guessed wrong.

**Describe the solution you'd like**
If you add mouseover text to the icons that explains what each slider/function does, it would remove any ambiguity without increasing GUI size or reducing elegance.

meteyou, 2020-11-12:
thx for your request. I planned to add a label in the next version. it isn't completely finished ATM, but I will post a screenshot ASAP. only icons are too less Info...

meteyou, 2020-11-25:
here is a screenshot from the next version. i display the names of each slicer now.

[image]

yonkiman, 2020-11-25:
Sweet!  Thanks again for making mainsail (and making it so great)!

meteyou, 2020-11-25:
i only have to fix some restart bugs with the temp chart... then i will
publish the next version. I printed some days with the new version and I
like it very much!

closed 2020-12-10 by meteyou