mainsail-crew/mainsail #129: [FR] Emergency stop button location too close to browser menu on mobile.

cbc02009, 2020-11-27:
**Is your feature request related to a problem? Please describe.**
The emergency stop button on mobile is too close to the menu buttons on most mobile browsers, and it's really easy for someone like me to fat-finger an emergency stop when all they wanted to do was open a new browser tab.

**Describe the solution you'd like**
I honestly don't know where a better place would be. Top left corner maybe? Or have a confirmation popup on mobile devices? I wish I had better suggestions.

**Describe alternatives you've considered**
I could just use my laptop, where this wouldn't be an issue, but having mobile support for mainsail is one of my favorite features of it.

**Additional context**
None.

cbc02009, 2020-11-27:
Here's a screenshot to highlight the issue
[image]

meteyou, 2020-11-27:
there is already a FR... https://github.com/meteyou/mainsail/issues/114

closed 2020-11-27 by meteyou

cbc02009, 2020-11-27:
Oops. Sorry about that!