mainsail-crew/mainsail #131: Power module doesn’t show after Moonraker update

LazaroFilm, 2020-11-28:
**Describe the bug:**\
A clear and concise description of what the bug is.
Moonraker changed how Power on/off is handled and broke the power module in Mainsail. 
**Expected behavior:**\
A clear and concise description of what you expected to happen.
See the power module when it's configured. 
**Screenshots:**\
If applicable, add screenshots to help explain your problem.

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
- klippy.log
- moonraker.log

lixxbox, 2020-11-28:
Should work again in the coming update.

meteyou, 2020-11-28:
@LazaroFilm pls try the new release and report...

LazaroFilm, 2020-12-10:
Works since update

closed 2020-12-10 by LazaroFilm