mainsail-crew/mainsail #133: No preview gcode file thumbnails since 0.3.0

nklrk, 2020-11-28:
**Describe the bug:**\
Since version 0.3.0 there is no preview thumbnail on the gcode file tab anymore. I don't know if this is a bug or a feature. But neither in the grid nor when clicking on a file there is no preview image shown. 

**Expected behavior:**\
There was a preview image earlier.

**Screenshots:**\
[image]
[image]

meteyou, 2020-11-28:
do you update the moonraker 
dependencies?

nklrk, 2020-11-28:
Yes I did. Mainsail shows me "v0.9.1-23-g41c6b4b3-dirty"

nklrk, 2020-11-28:
Ok please excuse me. Obviously there went something wrong while updating. I run the dependency update again, restarted the PI twice and now everything's fine again. Sorry for that 👍

closed 2020-11-28 by nklrk