mainsail-crew/mainsail #136: [FR] Use the klipper simulator to predict print times

zbrozek, 2020-12-05:
I'd love it if every gcode file got run through the klipper simulator and a total print time determined. It'd be great to have that displayed while printing, and maybe compared with a current run time.

Do a thing and wait commands are obviously a problem with this approach, so truncating heatup gcode prior to running through the simulator may be needed?

meteyou, 2020-12-05:
ATM there is no klipper simulator available for this feature. please make a FR in the klipper github.

but i talked already with Kevin and Eric about a simulation, but it needs a lot of performance...

zbrozek, 2020-12-05:
Ah, right, that's not an option because you're not executing code on the host. I could totally imagine running the simulator on gcode on upload in the background at low priority and then caching the result.

closed 2020-12-05 by zbrozek