mainsail-crew/mainsail #137: [FR] Second vertical scale on the temperature plot

zbrozek, 2020-12-05:
Bed and tool are usually quite-different temperatures. It'd be great to have better fidelity on the bed even when the tool is hot by having two different vertical scales. Left scale could be hotend, right scale could be bed. Bonus points for setting the bounds at printer.cfg-defined min and max plus a small margin. Super bonus points if there's a way to visualize how close the current status is to triggering validators.

meteyou, 2020-12-05:
thx for the FR. this would be possible! i will try it, but i would only make this feature optional, because i think most users are used to use absolute charts.

zbrozek, 2020-12-05:
Well it would still be absolute (reference to zero, or some klipper-determined minimum on the bottom), just that there would be two scales. If you wanted to make it not absolute, fixing the setpoint to somewhere mid-scale might be an interesting thing to play with.

meteyou, 2020-12-05:
yes, but not between extruder and heater bed, it is not absolute. but let me try something..

meteyou, 2020-12-27:
we tested a second y axis and it is a very bad usabillity solution. sry...

closed 2020-12-27 by meteyou

ushiocheng, 2024-02-21:
I am monitoring motor temps + chamber temp + mcu temp which means 7/8 of my temperatures are below ~100c so have a individual axis for tool would be great.
My compromise/workaround is just hide hotend temp and turn on auto scaling. Hope this helps.