mainsail-crew/mainsail #139: [FR] Remove or rename "save config" button in the top right

zbrozek, 2020-12-05:
The "save config" button is ambiguous. What config is it saving? Turns out it's the klipper config, and this is basically something I never want to push. It interrupts an ongoing print because klipper restarts, yet looks like a button that you're supposed to push (blue, has the word save in it).

In practice I never want to push that button. I suspect the reason that it appears is that mesh leveling has executed and there's an in-memory bed mesh. But saving that to the config makes no sense to me, so I don't even know why Klipper has that feature. I deliberately clear it on homing anyway.

meteyou, 2020-12-05:
SAVE_CONFIG is the name of the klipper command. i will deactivate it during a print, but not rename it.

nklrk, 2020-12-05:
This is a great idea. I use bed mesh before every print. Klipper writes this default mesh in printer config file. So I see the "Save Config" every print even I never save this mesh. So disabling the button during a print is a good approach 👍🏻

meteyou, 2020-12-27:
pls test the new release. the save_config button is now disabled during a print.

closed 2021-01-20 by meteyou