mainsail-crew/mainsail #141: [FR] / [Q] FFMPEG and rtsp://

le3tspeak, 2020-12-09:
Hi, is it possible to integrate an IP camera stream in Mainsail with FFMEPG via rtsp: // links?

meteyou, 2020-12-10:
thx for the FR. it is on my backlog and it's a very simple feature. maybe i can add it to the next release.

lixxbox, 2022-01-17:
Browser compatibility of rtsp-streams is apparently a bit "tricky". However, we are still working on our small side project "crowsnest" and the inclusion of WebRTC streams in Mainsail. With this it will most likely be possible to convert rtsp streams.

Unfortunately, the feature is dependent on crowsnest, which is why I set the status to "blocked" for now. We will keep you up to date.

https://github.com/mainsail-crew/crowsnest

meteyou, 2023-03-13:
i will close this FR, because we have implemented HLS and WebRTC streams in Mainsail.

closed 2023-03-13 by meteyou