mainsail-crew/mainsail #143: Macros not displayed after firmware restart until browser refresh

bung69, 2020-12-13:
Spend a while chasing my Tail on this one. 

initially thought macros were out right not being detected after firmware restart.  turns out a browser refresh is required after a firmware restart for the macro buttons to be displayed both in the dashboard and under settings > interface

Clean klipper Log
https://pastebin.com/5BwgHkD9
Clean Moonraker log
https://pastebin.com/rWx0LPg0

meteyou, 2020-12-15:
pls upload logfiles with the issue in it. i think this should be the same issue i fixed yet -> https://github.com/meteyou/mainsail/commit/bca27fadb703602653ae754b0fd6ec04c18dc62d

should be in the next release. pls test it if i publish it...

meteyou, 2020-12-27:
pls test it again. should be fixed in the new release.

closed 2021-01-20 by meteyou

GordonKI, 2024-08-01:
Just had the same issue with latest update.. anybody else having this issue