mainsail-crew/mainsail #144: [FR] Input fields and confirmation options for macro buttons

bung69, 2020-12-13:
I have a tool changing printer.  To handle unclean shut downs / Estops etc I have a macro [SET_CURRENT_TOOL]  that is used with a variable T= to manually set the currently attached tool.  this allows safe homing of the printer, and allows the printer to dock / unlock the tool in the correct place without crashes etc.  

In Octoprint I used a custom button with a input field to call this macro.  it also had a confirmation dialog to prevent accidental use while printing.  

Another use case for input fields would be to tweak offsets

**Describe the solution you'd like**
under settings > interface > macros each macro should have an input field allowing the user to specify Variables that if defined display an input field box next to the macro on the Dashboard. 

A simple checkbox allowing a confirmation dialog per macro would also be great. 

basically replicating the functionality of Custom controls in Octoprint

**Describe alternatives you've considered**
Having a boat load of macros, [SET_CURRRENT_TOOL_T0] [SET_CURRRENT_TOOL_T1] etc though this is not a solution for setting offsets etc.

meteyou, 2020-12-13:
it's on my backlog but not so important ATM for me.

fyi Klipper can safe variables in a file now. this is for tool changer and other special setups to safe data like this. it's called save_variable

bung69, 2020-12-14:
Ok cool, good to know. I may attempt it myself but probably not for a while as i should probably move all my tool change macros to python extension for klipper first. 

yes I noticed the ability to save variables to a file. should be useful for updating tool offsets and I may use them to save the current tool before user initiated firmware restarts.

I guess saving the current tool variable after every tool change would cover any Estops but doesn't cover me if i have manually changed a tool and forgotten about it.

crashes are fairly dramatic when changing tools at 1500mm/s and 10m/s/s so the more idiot checks the better.

closed 2020-12-14 by bung69