mainsail-crew/mainsail #145: Temp graph drops out during print at 238c on Nozzle and 70c on bed

MalsakerLures, 2020-12-15:
My temp graph in Mainsail drops out during print at 238c on Nozzle and 70c on bed.
This does not happen when printing at lower temp around 235c and 65c.

There is no drop in actual temperature in either Nozzle or Bed and the print continous. When refreshing the webpage its works for a period of time.

Unable to add Moonraker.log as it in on 33mb
Klippy.log: https://pastebin.pl/view/c5b332be

[image]

meteyou, 2020-12-15:
i changed the temp chart. pls test it again with the next release.

MalsakerLures, 2020-12-15:
shall i update to the latest mainsail and moonraker?

meteyou, 2020-12-17:
i will ping you, when i publish the next release.

MalsakerLures, 2020-12-18:
Perfect! Thank you!

Skickat från min iPhone

meteyou, 2020-12-27:
new release is published. pls test it again.

MalsakerLures, 2020-12-29:
Seems to be working now👍👍

Skickat från min iPhone

meteyou, 2020-12-30:
thx for the report!

closed 2020-12-30 by meteyou