mainsail-crew/mainsail #146: Buttons get stuck in spinning state until page refresh

bcbuckles, 2020-12-17:
Buttons get stock in "spinning" loop if the macro or action cant be completed, a page refresh must be done before they become accessible again.  For example, clicking QGL, console returns that axis must be homed first.  QGL button is not stuck in the "spinning" state and cant be clicked.  You then need to home, then refresh, then click QGL again.  This appears to be the case for all buttons, homing, qgl, macros, etc.

meteyou, 2020-12-17:
i already fixed it -> https://github.com/meteyou/mainsail/commit/bca27fadb703602653ae754b0fd6ec04c18dc62d

is in the next release

closed 2020-12-18 by meteyou

JamesIV4, 2022-08-01:
I'm seeing this bug in the latest version of Mainsail (2.2.1).