mainsail-crew/mainsail #147: [FR] Srubber for values on plots

bcbuckles, 2020-12-18:
Would be nice to be able to scrub over the plot and get a vertical line, and readout of all temperature values at that time value.  Like in Octoprint or other plotting libraries.  I often use this to look at trends in chamber temperature that don't show up well due to their small deltas that are less noticeable on the scale of Toolhead temp.

meteyou, 2020-12-18:
yes. thats the reason, why i changed the chart libary. will come...

bcbuckles, 2020-12-19:
> yes. thats the reason, why i changed the chart libary. will come...

Great news.  What's protocol here, should I delete the FR?

meteyou, 2020-12-19:
No. Let it open, so others users can see, that it is a open feature. I will close it, if I fixed it.

meteyou, 2021-01-30:
i added this feature. will be in the next release:

[image]

closed 2021-01-31 by meteyou