mainsail-crew/mainsail #149: Cannot delete files which contain the '+' character in the file name.

nazzrim, 2020-12-19:
**Describe the bug:**\
It is not possible to delete files via the mainsail UI if the filename contains the '+' character. After the file was renamed (via mainsail UI) it is working.

**Expected behavior:**\
Delete files which contain '+' in the file name.

**Screenshots:**\
[image]
[image]

**Logfiles:**\
[klippy.log](https://pastebin.com/kh2CkEhZ)
[moonraker.log](https://pastebin.com/fd51vb2x)

meteyou, 2020-12-19:
I cannot open your logfiles. pls upload the Moonraker log again.

nazzrim, 2020-12-20:
My bad. [Here](https://pastebin.com/0KUBUzyp) is the moonraker.log

commit d74e38d0 (2020-12-30): bugfix: delete files with special chars (fix #149)
closed 2020-12-30 by meteyou