mainsail-crew/mainsail #151: Latest release incompatible with Klipper

nleehone, 2020-12-27:
**Describe the bug:**\
After installing mainsail and setting up my configuration from Klipper I got the following error message:
NoOptionError: No option 'microsteps' in section: 'tmc2209 stepper_x'
See issue raised here: https://github.com/KevinOConnor/klipper/issues/3684

It turned out this was because they updated some of the configuration settings on 20201222: https://www.klipper3d.org/Config_Changes.html

Users installing from Mainsail might be confused because of the incompatibility between the latest config files, and the 0.9.1 version that Mainsail is bundled with.

**Expected behavior:**\
A working installation.

**Possible fixes:**\
Either make a new release with the new Klipper version bundled, or make it clear in the README that a user should update before using any of the config files.

meteyou, 2020-12-27:
I am finishing the next mainsail release and then ray will build a nice mainsailOS image with the new Klipper Version. the next mainsail release have a integrated update manager to update Klipper, moonraker, mainsail and the os packages. the release will be released in a few days.

meteyou, 2020-12-27:
release is published...

closed 2020-12-30 by meteyou