mainsail-crew/mainsail #154: High GPU usage even when mainsail browser window is minimized

cbc02009, 2020-12-30:
**Describe the bug:**\
High GPU usage by mainsail browser window, even when the browser window is minimized: 
[image]

from Firefox taskmanager:
[image]

**Expected behavior:**\
I would expect the browser to not use many resources when minimized or hidden, so that I can keep it open on my laptop for emergencies, but not have it kill my laptop's battery.

If this should be a feature request instead of a bug report, please let me know, and I'll change it.

meteyou, 2020-12-30:
pls send me your logfiles and tell me witch version do you use.

cbc02009, 2020-12-30:
I'm using version v0.3.0

Here's the moonraker and klippy logs:
[moonraker.log](https://github.com/meteyou/mainsail/files/5755479/moonraker.log)
[klippy.log](https://github.com/meteyou/mainsail/files/5755480/klippy.log)

meteyou, 2020-12-30:
ah ok. its the temp chart. pls update to 0.4.0. i changed the tempchart because of performace problems

cbc02009, 2020-12-30:
Should have checked for an update first...

I'll close this if that fixes this. Thanks for the help.

cbc02009, 2021-01-01:
@meteyou So that fixed half the problem. It definitely reduced the GPU usage from ~25% to ~1%, but it still uses ~4% cpu even when minimized or hidden, which has a drastic effect on my battery life.

I know literally nothing about web page design. Is it possible to stop drawing the graphs when the page is hidden or minimized (or just not the focus window)?

meteyou, 2021-01-02:
let me check this performance... but i think is is only the websocket, if you minimice or hidden the window and this is necessary for mainsail. i think the graph is not drawing/rendering, if it is hidden...

cbc02009, 2021-01-03:
I don't know if this is helpful, because the processor spikes make the averages hard to read, but here's my power usage on my desktop PC:

[image]

The first part is with the window open and showing, consuming ~105w.

The second part is with the window open and minimized, consuming ~100w.

The third is with mainsail window closed, consuming ~90w.

meteyou, 2021-02-02:
do you tested a newer release again?

wormiedk, 2021-02-03:
Mine uses around 10% having mainsail open even if minimized / closed and if I hide both the temp graph and camera. Firefox drops to around 2% when closing the mainsail tab

meteyou, 2021-02-05:
i cannot put mainsail in to a "sleep mode". otherwise a have to reinitial the while gui. its like a close and new opening of the while interface.

closed 2021-02-05 by meteyou

jamiehs, 2021-05-19:
I'm seeing this same issue again.
High CPU usage when the window is in the background or minimized (or the display turns off). The only way I know it's happening is because I have aggressive fan curves on this (gaming) machine and I can hear them ramp up.

When I look at the Mainsail window, interactivity is sluggish, the webcam module is running at 1fps, and a refresh fixes the issue. I hear the fans instantly slow down, and the fps returns to a nominal 15-16 fps. I also observed this before adding a webcam to my setup, so I don't suspect it has much to do with the addition of the cam.

* Windows 10
* Chrome 90.0.4430.212 (Official Build) (64-bit)
* Mainsail 1.6.0

[image]

jamiehs, 2021-05-20:
Looked into this a bit more. It looks like it was related to the camera.
The Adaptive MJPEG-Streamer seems to be what is causing the issue. I switched to the regular MJPEG-Streamer and it's doing much better.

Sorry for the extra noise in this thread, but hopefully it could help someone else who is having issues with high CPU and memory use.