mainsail-crew/mainsail #155: [FR] Optional / Alternate Jogging Controls

dspears0809, 2020-12-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

An optional, alternate directional control panel that has a layout similar to what's used in Fluidd and OctoPrint with the centered home button, directional buttons, and movement distance (configurable movement distance selections which is already requested in Issue #134).

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

meteyou, 2020-12-31:
thx for the FR. I've been planning for a long time, but I haven't time to it yet.  please let the FR open until I've done it.

meteyou, 2021-03-11:
@dspears0809 pls test the new release (v1.4.0). there is an alternative control option.

closed 2021-03-15 by meteyou