mainsail-crew/mainsail #156: What is minimum system requirements for mainsail ?[FR]

furkanbr, 2021-01-02:
Hi everyone ! 

I think this is a great project and want to use on my printers. One of my printers have RPi 4 and im using that with octoprint but otherone dont have anythink on it.  my question is , i have another Raspberry pi zero that i dont use right now , can Raspberry pi zero run mainsail and klipper firmware? do i get any issue with that setup? 

Thanks for this open source project , Have a great day !!

meteyou, 2021-01-02:
i use also a pi zero in my switchwire. klipper, moonraker + mainsail works perfekt on it, the only problem is a webcam. i think the main problem for it its the single core processor of the pi zero, but if you dont use a webcam, feel free to us it.

furkanbr, 2021-01-02:
Thank you , i dont interest with cam features , i  will give a try!

closed 2021-01-10 by meteyou