mainsail-crew/mainsail #157: Wrong/old thumbnails are displayed

Metti93, 2021-01-02:
**Describe the bug:**\
In the main GUI, after a successful print where a thumbnail was included, the exact same thumbnail is also displayed if you start G-Code without a thumbnail.
See screenshots

**Expected behavior:**\
Load G-code from Superslicer, with thumbnail, into Mainsail.
Start print and thumbnail is displayed correctly.

Load second G-code, without thumbnail, into mainsail.
The G-code did not display a thumbnail, as it should.
Start the second print and the thumbnail of the first print is displayed.

**Screenshots:**\
[image]

[image]

**Logfiles:**\
[klippy.log](https://github.com/meteyou/mainsail/files/5760885/klippy.log)
[moonraker.log](https://github.com/meteyou/mainsail/files/5760886/moonraker.log)

eriroh, 2021-01-26:
I am experiencing the same thing and have not yet figured out a workaround except clicking (X) on the print.

meteyou, 2021-01-26:
@eriroh you only have to refresh your browser tab.

meteyou, 2021-01-31:
fixed with https://github.com/meteyou/mainsail/commit/e2e496c31e2995e220d22f67b3785feaea100d29

closed 2021-01-31 by meteyou