mainsail-crew/mainsail #158: [FR] adjusting feedrates

jtrmal, 2021-01-06:
**Is your feature request related to a problem? Please describe.**
I'd like to be able to change the feedrates -- even for HF dragon and 0.4mm nozzle, the values over 13mm/s are just too much...

**Describe the solution you'd like**
Perhaps manual configuration of the feedrates or some additional tool for covering a specific interval of feed rates using the maximum volumetric flow?

**Describe alternatives you've considered**
I changed them in the js file (yes I know that implies I won't be able to update automatically)

**Additional context**
IMO probably a low priority, but perhaps useful?

alexhocks, 2021-01-31:
+1 for this. My V6 can't handle anything more than 5mm/s so configuring the feed rates on the dashboard myself would be really nice.

closed 2021-02-07 by meteyou

jtrmal, 2021-02-07:
> Closed #158 <https://github.com/meteyou/mainsail/issues/158> via 3749b76

thank you!
y.

commit 3749b761 (2021-02-07): feature: add customize feedrate & feed distances to ExtruderPanel.vue (fix #158)