mainsail-crew/mainsail #159: Cannot upload files when Mainsail when the printer cannot communicate with the microcontroller

nightwolfix, 2021-01-07:
**Describe the bug:**\
If I try to upload a file or create a folder while mainsail is not connected to the microcontroller, then I get an error

**Expected behavior:**\
I should be able to upload file at any time

**Screenshots:**\
Red alert in the bottom right related to "gcodes"

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
- klippy.log
- moonraker.log

nightwolfix, 2021-01-07:
[moonraker.log](https://github.com/meteyou/mainsail/files/5781332/moonraker.log)
[klippy (1).log](https://github.com/meteyou/mainsail/files/5781333/klippy.1.log)

meteyou, 2021-01-10:
yes, because moonraker dont know where the virtual_sdcard directory is. i only can disable the gcode-files, when klipper is not running. is this ok?

closed 2021-01-31 by meteyou

davidhiendl, 2021-10-01:
It would be nice if g-code upload would be possible anyway, perhaps requiring manual input of the path via config menu (or parsing the printer.cfg). This would for example allow uploading g-code files before moving to a remote workplace and starting the printer.

meteyou, 2021-10-02:
yep, but this have to be a function of moonraker. mainsail just provide the upload "formular" to moonraker and only display it, when moonraker have the gcode directory. moonraker needs one start of klipper to parse the printer.cfg to know where this directory is (atm). parsing printer.cfg is not as easy as it sounds, because you can have included files (like mainsail.cfg in the case of mainsailOS). so you have to parse the full printer.cfg with alle dependences to know where the virtual_sdcard is.

davidhiendl, 2021-10-02:
Ah yes, Mainsail is client-side only and relies on moonraker for all IO, forgot about that.