mainsail-crew/mainsail #161: [FR] Add "home" button to the mesh bed

jtrmal, 2021-01-09:
**Is your feature request related to a problem? Please describe.**
It would be good if there was a homing button next to the calibrate button on the "mesh" page. I find it a bit annoying to have to switch the page to "dashboard" (or "console") and home and then switch back to "mesh"

**Describe the solution you'd like**
To have a button there that calls homing.

**Describe alternatives you've considered**
Modification of the homing macro to contain G28

**Additional context**
Add any other context or screenshots about the feature request here.

meteyou, 2021-01-10:
i recommend to adjust your bed_mesh_calibrate gcode (like this: https://docs.mainsail.app/necessary-configuration#customize-klipper-default-commands) but i will add a home button.

jtrmal, 2021-01-10:
thank you! Shall I close?

meteyou, 2021-01-10:
no. pls let it open. i will close it, when i implement this feature.

meteyou, 2021-01-20:
pls check the new release. i added a home button in the heightmap panel: https://github.com/meteyou/mainsail/commit/fe246393ea652c21afa4c1cf7db0fa8c9511a732

jtrmal, 2021-01-20:
> fe24639

yes, looks great -- already checked that out yesterday. Thanks a lot!
y.

closed 2021-01-23 by meteyou