mainsail-crew/mainsail #164: Update button no feedback

barecool, 2021-01-11:
**Describe the bug:**\
If printer is active (printing) update panel should be disabled
mainsail allows to press the update button, however no feedback is provided

**Expected behavior:**\
Panel should disable while printing / or provide a toast with feedback that updating is disabled while printing

**Screenshots:**\
[image]

**Logfiles:**\
[klippy.log](https://github.com/meteyou/mainsail/files/5796646/klippy.log)
[moonraker.log](https://github.com/meteyou/mainsail/files/5796650/moonraker.log)

meteyou, 2021-01-11:
yes, this is a new moonraker feature. i have to implement it in mainsail... pls let this ticket open. i will close it, when i fixed it.

meteyou, 2021-01-19:
i fixed it in https://github.com/meteyou/mainsail/commit/d05fa70832efc91b7a9a39384fac1c8665edc543. thx!

closed 2021-01-19 by meteyou