mainsail-crew/mainsail #169: [FR] Exclude Region

stefanpieter, 2021-01-18:
The exclude region is one feature I really miss about Octoprint. I didn't see it in the list so wanted to get it out there.

I am aware this is a very difficult feature to add however is is awesome. Especially when doing full plate prints.

meteyou, 2021-01-19:
Object/region canceling is on of the most powerful functions i also like to integrate in mainsail, but the first issue is, that mainsail is only a interface and dont read/render the gcode. only klipper "render" the gcode and can skip gcodes during a print. thats a big difference between a webclient behind moonraker and octoprint. octoprint is a host software and only "stream" gcode. for example mainsail i only a webinterface to display the current state from klipper. Here is the current FR in klipper for this feature -> https://github.com/KevinOConnor/klipper/issues/3375

stefanpieter, 2021-01-19:
Thank you Stefan

I am sure we will get there eventually and I look forward to that day ;-)

Thank you for all the awesome work!

closed 2021-01-19 by stefanpieter

watzon, 2022-06-22:
Seeing as `exclude_object` has been added to Klipper, now might be a good time to reopen this FR. Something that would be really nice is if the GCODE viewer had this functionality built in. I'd love to be able to just right click and select `Exclude` or something.