mainsail-crew/mainsail #170: [FR] Add Upload and Start print button on the dashboard

DorjeDorf, 2021-01-19:
**Is your feature request related to a problem? Please describe.**
It's a usability feature I think. The way I see the use of Mainsail now, I have to go the Gcodes tab, upload a file, then click it to start printing. 

**Describe the solution you'd like**
Ideally, we'd see a button next to the Emergency Stop that would allow us to browse to a file, have it uploaded and print start automatically. Total rip off from DWC, but it does speed things up a lot ;)

**Describe alternatives you've considered**
None. Nice usability feature. 

**Additional context**
Nothing to add - basic feature in DWC. Would make Mainsail even more convenient to use.

meteyou, 2021-01-20:
Ohh... You feature request was close before I released v1.0.0. pls update and enjoy the "upload & start" button in the topbar.

DorjeDorf, 2021-01-20:
Now that's what I call amazing :)

Thanks, works perfectly!

closed 2021-01-20 by DorjeDorf