mainsail-crew/mainsail #171: Thumbnails are not displayed after update to 1.0

toskium, 2021-01-20:
**Describe the bug:**\
In prior version 0.4.1 thumbnails have been displayed.
After update of mainsail to version 1.0.0 thumbnails don't appear.

**Expected behavior:**\
Thumbnails should appear.

**Screenshots:**\

[image]

pierre-vo, 2021-01-20:
`bugfix
display only 32x32 thumbnails in gcode files`

I think it is just more strict on the size of the small thumbnails. I had 64x64 and it was working, I expect it not to work anymore after the update from this part of the release note and will update my slicer to use 32x32.

Not sure though, I have not tested yet.

toskium, 2021-01-20:
Thanks for the assist, 32x32 seems to be the only size that works. Too bad, I liked the big thumbs :-)

meteyou, 2021-01-23:
it was only a bug. 32x32 is the correct size. i added a hover for the next version.

closed 2021-01-23 by meteyou