mainsail-crew/mainsail #175: [FR] plot heater status (PWM value)

dewi-ny-je, 2021-01-22:
When I heat up the bed the desired temperature reaches the set value much earlier than the pwm value supplied to the heater stabilises, since it takes time for the heated bed to soak up the heat, especially when a glass plate is used on top of the aluminium bed.

Right now I look at the status value but depending on the set temperature the final steady state value is different and I don't remember all of them. Also, sometimes I print enclosed, sometimes not, so for the same set temperature the final steady state pwm value changes.

I think it would be very useful, for some users, to plot the pwm value together with the temperature graph, using obviously a second vertical scale which goes 0-100%.

This way I would be able to see with a quick glance whether the own value is stable or it's still decreasing.

This is especially useful for users with PID controller, but also users with bang-bang control would see whether the on/off intervals are equally spaced or still getting more rarified.

meteyou, 2021-01-23:
i think that there is also a wrong result. i use 8mm aluminium plates and until they heat up, i can't even orientate myself to the pwm values. the bed heats up much faster at the bottom centre (where the sensor is) than on the top side up to the edges. the best variant for something like this is a second thermistor.

i will try it anyway to see how it would look. i just think that the pwm jumps too much and only one bar would be visible. i will post a screenshot, when i test it...

dewi-ny-je, 2021-01-24:
Mine is quite consistent and it jumps only about 4% when it reaches equilibrium, but it's also true that I have a zero-crossing SSR for a 230V heater so the pwm resolution is 50 cycles/s *0.3 s = 15 power levels, which is 7% power steps

meteyou, 2021-01-27:
i tested it and it will be a part of the next release!

[image]

dewi-ny-je, 2021-01-27:
I don't know CanvasJS but if you want a nice UI I suggest putting labels on the vertical scales: PWM on one side and °C on the other side.
Pwm of course is wrong, since that axis has no unit of measure, but something is needed.

meteyou, 2021-01-29:
thx for the feedback. i added units to both axis.

[image]

closed 2021-01-31 by meteyou