mainsail-crew/mainsail #177: [FR] add support for Cura's ufp file format with thumbnail

TimoErdmann, 2021-01-27:
The ufp File Format from Ultimaker / Cura is a zipped container which contains a screenshot and a compressed version of the gcode. It safes more than 2/3 of space compared to a gcode file and it contains a neat thumbnail. 

For Octoprint exists already a plugin: https://plugins.octoprint.org/plugins/UltimakerFormatPackage/

that plugin above can show the thumbnail and the following one is needed to pring the gcode which is packed in the container:
https://github.com/fieldOfView/Cura-OctoPrintPlugin

meteyou, 2021-01-27:
moonraker already support ufp files. there was only a bug in the current release, that ufp files was not allowed. here is the bugfix: https://github.com/meteyou/mainsail/commit/e5469b47caf5c2e4fe7c5ae0b166e810f48ad00c .

will be included in the next release. thx for the FR.

[image]

TimoErdmann, 2021-01-27:
oh thats great! Thank you for your work!

closed 2021-01-31 by meteyou