mainsail-crew/mainsail #180: [FR] Support for thumbnails from Ideamaker

modmt, 2021-01-31:
Hello,

is it possible to display thumbnails from the Slicer Ideamaker? That would be a nice feature for me, because I switched to Ideamaker from S3D and Prusaslicer. As I understand only Prusaslicer and Cura is supported at the moment regarding Thumbnails.

Thank you and best regards,
Matthias

meteyou, 2021-01-31:
prusaslicer & cura writes the thumbnail into the file. does ideamaker do that too? if ideamaker does that too, moonraker could extract it and i could use it in mainsail. but the extracting itself happens in moonraker.

modmt, 2021-01-31:
As I understood there is the possibility in Ideamaker to wrote Out an additional .Data File, which contains a Picture for displaying the model on the Raise3D Machines. But I do not know how this could be used.

I will ask this tomorrow in the moonraker git.
Thank you so far for your quick reply Stefan and Thank you too for your really good work with Mainsail.

meteyou, 2021-02-22:
@modmt do you have any news for me?

modmt, 2021-02-23:
Hello Stefan, unfortunatelly no update at the moment. I checked the .data file of Ideamaker but it seems to be that there is no easy way to extract this or at least I do not know how.
So I did not ask in the moonraker git for that until today. 
But I think I will do this this evening, maybe he knows a solution. I will give an update if I know more.

meteyou, 2021-03-17:
are there any news about ideamaker thumbnails?

closed 2021-04-20 by meteyou

julianobst, 2021-10-20:
Moonraker was up-to-date but I had no square format of the thumbnail set in IdeaMaker.
Setting the resolution to 400x400 (as you said) did the trick.

I guess this important information should be added to the mainsail docs: https://docs.mainsail.xyz/quicktips/thumbnails

Thanks for the fast reply and solution.