mainsail-crew/mainsail #181: update manger / updating Moonraker fails

wuestner, 2021-01-31:
**Describe the bug:**\
A clear and concise description of what the bug is.
In the update-manager, moonraker shows the status "dirty".
Updating results in an error message: "Update aborted, repo is has been modified"

**Expected behavior:**\
A clear and concise description of what you expected to happen.
Update if moonraker in GUI is possible.
**Screenshots:**\
If applicable, add screenshots to help explain your problem.
[image]

[image]
**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.

https://pastebin.com/bmZ05XpY

meteyou, 2021-02-02:
"dirty" means, you have changed something in this directory. pls don't touch the klipper or the moonraker direktory. you can fix it with:

```
cd ~/moonraker
git reset --hard
```

wuestner, 2021-02-02:
`pi@mainsailos:~/moonraker $ git reset --hard
HEAD is now at e2c325a extract_metadata:  support first layer height expressed a                                                                                                                                                           s a percentage`

mainsail still shows "dirty" in the gui

closed 2021-02-02 by meteyou