mainsail-crew/mainsail #183: [FR] Three columns on dashboard and repositioning dashboard items

alexhocks, 2021-01-31:
**Describe the solution you'd like**
On larger monitors, such as 2k resolution and above, I feel there is room on the dashboard for a third column of panels. Especially with the menu collapsed, there definetely is enough space.
Combined with the possibility to move dashboard panels around, this would mean more freedom for personalized setups.
The god tier for this feature would be to make the panels be configurable such that the user can choose how many columns a panel should span.

[image]

meteyou, 2021-02-02:
atm i focus on support all klipper features. something like gui customizing with move panels have a very low prio for me.

meteyou, 2021-08-03:
Pls check the new V2.0.0-beta2 release... i added a function for this...

closed 2021-08-31 by meteyou