mainsail-crew/mainsail #185: streaming from ESP32 cam

robvoi, 2021-02-02:
**Describe the bug:**\
I use an ESP32 cam as webcam. I can stream in the browser via http://http://192.168.1.165:81/stream
If I enter this URL as webcam URL I get no picture in Mainsail.

May be related to: #141 
In case streaming doesn't work this way yet.

meteyou, 2021-02-02:
esp32 cam streams mjpg. pls only type 1 time http:// in the url field. this is the protocoll. try `http://192.168.1.165:81/stream`

robvoi, 2021-02-03:
> esp32 cam streams mjpg. pls only type 1 time http:// in the url field. this is the protocoll. try `http://192.168.1.165:81/stream`

Thanks for your response. So it at least is supposed to work. 

The double http:// was a copy/paste mistake. 
I put the correct URL in the config. But I don’t get the stream in mainsail. With the same URL I get the stream in a blank tab.

Do you have another idea in what might be wrong?

commit 837f6641 (2021-02-06): bugfix: esp32 cam or ustreamer link fix bypassCache append (fixed #185)
closed 2021-02-06 by meteyou