mainsail-crew/mainsail #186: Backup of printer.cfg to gdrive/ftpserver [FR]

kaisertheone, 2021-02-02:
in the last few days i lost the printer.cfg file during a corrupted pi system.
i didnt had a recent backup transfered to another system.

i would wish for the possibility to set a server (ftp) maybe a storage in the network or a googledrive document and mainsail uploads the printer.cfg on every boot.

zellneralex, 2021-02-03:
Use rsnapshot  <https://rsnapshot.org/> as destination you can do everything you want. I have a samba running that connects my NAS.

th33xitus, 2021-02-03:
How often do you change stuff in your printer.cfg that you need an upload on every boot?
You could use github too. Shell script, cronjob... push it there...
There are so many solutions for backing up files somewhere, i don't think mainsail should be tasked with it.

meteyou, 2021-02-05:
@kaisertheone mainsail is only execute in your browser. there is no running task on the host/rpi for create a backup or anything else. if i would implement a backup function in mainsail, it should download all data from the rpi to your local pc and then backup it. i don't think this will work well. but you can ask moonraker for implement a backup feature. furthermore there are enough free/opensource tools for linux to set up such a backup system...

kaisertheone, 2021-02-05:
i think under this circumstances this FR can be closed :)

closed 2021-02-09 by meteyou