mainsail-crew/mainsail #189: Temperature presets appearing after filling in the temperature manually.

MarcPot, 2021-02-04:
When manually filling in a temperature, the preset list is acting unexpected.

First when clicking the temp it appears (expected behaviour).
[image]
Then when typing in the temperature it disappears (expected behaviour).
[image]
But then when pressing enter because you filled it in, it appears again (unexpected behaviour).
[image]

The reason why it's annoying is because the preset bar overlaps the other temperature textboxes, so I first have to click somewhere random before being able the continue.

DruckiMcDruck, 2021-02-04:
I also have this issue, i'm running Mainsail v1.1.0.

meteyou, 2021-02-05:
thats the way, how a combobox work. its a input field and a select/dropdown box in one.

MarcPot, 2021-02-05:
Yes but my expectation would be that the dropdown would disappear after pressing enter.

If this is the standard behavior I’d prefer a simple input field. But I guess the dropdown box is useful for mobile users?

meteyou, 2021-02-05:
its the same if you set a value or you will change a value. if your printer is set to 200° and you will set it higher or lower with a click/touch in the field, you would need this values again. i cannot separate these 2 situations

MarcPot, 2021-02-05:
To bad. Thanks anyways!

closed 2021-02-05 by MarcPot

meteyou, 2021-02-05:
i will check it again, but at the moment i don't know a way out...

intelinc, 2022-05-16:
This is a perpetual issue for me on desktop and mobile. The nonstandard behavior is mainly that hitting enter should set the temp and remove the selections. It does the former, but not the latter, which requires an additional click outside of the temp combobox to fully "fix" the temp selected. Maybe this works well for dropdown selections, but I tend to work across a lot of filament types, so I just type the temp in. 

An additional problem with the dropdown remaining is that the extruder temp dropdown is almost visually identical to bed temp. 

inactive temp menu:
[image]

extruder temp dropdown superimposed over bed temp:
[image]

At a glance and on mobile, these are visually equivalent, so I am constantly typing in the extruder temp, pressing "enter" and then clicking the 0℃ dropdown when I mean to set the bed temp. So instead of setting the bed temp, I inadvertently set the extruder back to zero. Constantly. It's just muscle memory from a few years of running Fluidd (which does clear the dropdown on enter) before switching to Mainsail.

th33xitus, 2022-05-16:
@intelinc 
In the current beta release, we got rid of the combined dropdown. now a dropdown with presets will only open, if you click on a button with the caret down symbol. the temperature input field will not toggle a dropdown.
so once v2.2.0 is released you will not have this issue anymore. or you update now to the current v.2.2.0-beta3 and get rid of that issue immediately :P the choice is yours.

intelinc, 2022-05-16:
@th33xitus thanks for letting me know. Looking forward to trying this out!