mainsail-crew/mainsail #190: Temperature of extruder isn't applied when set manually during print start

DruckiMcDruck, 2021-02-04:
**Describe the bug:**\
When starting a print, Mainsail usually heats the bed first and only heats the extruder afterwards. When trying to set the extruder temp manually while the bed hasn't reached its target, the extruder temp is applied, but the extruder won't heat.

**Expected behavior:**\
If i set the extruder temp while the bed is heating up i expect Mainsail to heat both simultaneously. However Mainsail shows the new extruder target, but won't heat it. Note the timestamp of the entries in the console at the bottom. I set extruder temp at 21:42:59 and still the extruder temp drops further.

**Screenshots:**\

[image]

meteyou, 2021-02-05:
thats your start gcode. i think you use M190 (Set bed temperature and WAIT) in your start gcode. is klipper in the WAIT state, it cannot execute any gcode/command. you see the sending command in the console, but klipper will not execute it.

DruckiMcDruck, 2021-02-05:
Thanks for your hint. I investigated further and it seems like Cura defaults to M190 instead of M104/140. You need to set a specific start code to make cura use M140/104 instead... 

Works now, so closing here as this is intended behaviour and not a bug.

closed 2021-02-05 by DruckiMcDruck