mainsail-crew/mainsail #191: [FR] ARC support (replicate arc welder OctoPrint plugin)

robvoi, 2021-02-05:
**Describe the solution you'd like**
It would be great to have the arc welder feature from this plugin replicated for/in Mainsail.
https://plugins.octoprint.org/plugins/arc_welder/

DruckiMcDruck, 2021-02-05:
If you're a Cura User, you can just use the ArcWelder plugin. No need for support in Octoprint or Mainsail then ;)

meteyou, 2021-02-05:
mainsail is only a webinterface and have no gcode access. this have to be a a plugin on the host side (moonraker/klipper). i think someone have already work on a arc weller moonraker plugin.

otherwise, however, with mainsail it is not necessary to convert gcode to G2/3, since klipper can handle enough gcodes on its own without gcode filter.

th33xitus, 2021-02-05:
In the meantime you could download this: https://github.com/FormerLurker/ArcWelderLib/releases/tag/1.1.0
And just arcweld locally on your PC before uploading the gcode.
Its probably also possible to include this into some kind of post-processing script of a slicer to have it automated.

robvoi, 2021-02-08:
Thanks for all the responses.
@meteyou If there is indeed no practical advantage here, then indeed the request is obsolete.
@th33xitus Thanks. didn't see there is a standalone version of it. Only knew the plugin.

paeppi88, 2021-02-09:
Prusa Slicer also supports a direct application of the Arc Welder plugin. Just search for Arc Welder Prusa on YouTube :)

DruckiMcDruck, 2021-02-09:
There are currently four versions of ArcWelder that i know of:
1. Octoprint Plugin
2. Cura Plugin
3. Prusaslicer Plugin
4. Standalone

There's also a php based solution of another similar tool which you can host locally. However i prefer one the other four options mentioned above. The Cura plugin by fieldofview works great and processes every gcode automatically. You can even disable it or change its settings in the settings sidebar inside Cura. 

When using it, i noticed that my printer runs noticeably smoother when printing circles. Before using it you could hear the instant changes in direction when printing line segments of a circle. That's now gone and the change of motor speed is absolutely smooth.

meteyou, 2021-02-09:
thx guys for this informations! now i'm also triggered to test it with prusaslicer!

can we close this issue or is something todo?

robvoi, 2021-02-09:
If there are no plans for an implementation in mainsail (process on upload) then sure, please close.
There is a solution with the standalone version. Just not as comfortable (usinf IdeaMaker - no post processing with external programs).

In general - great work with mainsail, thanks. Keep it up. :-)

closed 2021-02-09 by robvoi

meteyou, 2021-02-09:
Sry... It's not the right place for it. Maybe it should be possible in Moonraker as a plugin.