mainsail-crew/mainsail #192: [FR] Add support for DHT22 temp sensors

DorjeDorf, 2021-02-05:
**Describe the solution you'd like**
There is a plugin in Octoprint called Enclosure. It has one feature in particular that I would love to see in mainsail and that is the support for temperature sensors connected to the raspberry pi. Is that something you could do within mainsail? 

Here's a link to how it's done for enclosure:
https://github.com/vitormhenrique/OctoPrint-Enclosure

Would be awesome for a boards where you don't have any more slots on the SKR, but could use GPIO of the raspberry.

meteyou, 2021-02-05:
mainsail is only a webinterface and is only running in your browser and have no connection to any hardware of your printer. it only display the state of your printer. pls make a FR in klipper for it.

FYI... i use a BME280, that is connected to the rpi for my chamber temp:
[image]

you can add your rpi as a mcu in your config and use all gpio pins: https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md

but pls dont make FR for hardware support. in mainsail it is not possible to support hardware devices. this has to happen on the host itself (klipper or moonraker).

theopensourcerer, 2021-02-06:
@DorjeDorf This is nothing to do with Mainsail. 

I also used to use the Enclosure plugin with a DS18B20 1-wire sensor, but have now moved to Mainsail. Recently myself and others developed support for the DS18B20 in Klipper and it is now a feature in the master branch [1]. You should check if there is any development or support for the DHT family and if not, start working on support for them :-)

The RPi is a fully supported "mcu" in Klipper so using GPIO on the RPi is fine already. You just need to get the communication between the RPi and the sensor working.

[1] https://www.klipper3d.org/Config_Reference.html#ds18b20-temperature-sensor

theopensourcerer, 2021-02-06:
@DorjeDorf It looks like someone has had the same idea already... ;-) 

https://github.com/KevinOConnor/klipper/issues/3886

DorjeDorf, 2021-02-06:
Thanks for the links. I'll check those out ;)

closed 2021-02-06 by DorjeDorf

Lubbert-Airsoft, 2022-04-04:
> mainsail is only a webinterface and is only running in your browser and have no connection to any hardware of your printer. it only display the state of your printer. pls make a FR in klipper for [...]

But how to? i try serval times on the I2C bus but not get it to work.
Some information of how it works i cant find a good skript like ADXL that works so nice.

meteyou, 2022-04-04:
@Lubbert-Airsoft pls ask on our discord in the klipper channel or use a klipper relevant platform...