mainsail-crew/mainsail #197: Raspberry Pi temperature color in chart not functioning

dreedijk, 2021-02-09:
When the Pi is added with (according to docs.mainsail.xyz):

[temperature_sensor raspberry_pi]
sensor_type: rpi_temperature
min_temp: 0
max_temp: 85

the chart is displaying a blue line indicating the temperature (default color). When changing this color, the color is not displayed. When going back to the color selection, the color is reset to the default value, but the line is no longer visible in the chart.

[image]

You would expect the line in the chart, to display in the selected color.

I previously added:

[temperature_sensor host]
sensor_type: rpi_temperature
min_temp: 0
max_temp: 85

This gave me the option to show/hide the temperature in the chart. With the [temperature_sensor raspberry_pi] option however, this is not an option.

If you need logfiles, please let me know. I have not added them for privacy reasons.

dreedijk, 2021-02-09:
Another thing I noticed. I thought I'd then change the heat bed color to at least distinguish between the temperatures in the chart and by changing the Heat bed color, also the Raspberry pi color changed to the same color. So it appears that the colors are linked ...

meteyou, 2021-02-09:
i fixed this already. will be in the next release -> https://github.com/meteyou/mainsail/commit/6fb8d1614357ba3a086d724d970dc20a2634da64

closed 2021-02-09 by meteyou