mainsail-crew/mainsail #204: Hight CPU usage and unusable on mobile

SilverWx, 2021-02-18:
[image]

I goit hight CPU usage on my computer, and interface seriously lag on my Galaxy S10.

I use brave webbrowser bassed on chromium both on my pc and my phone.

meteyou, 2021-02-18:
mainsail is a dynamic gui and data is streamed from moonraker in 200ms intervals! the consumption MUST be higher than with a static website.

SilverWx, 2021-02-18:
Ok I understand that bu i'm not experiencing this lag it's fluidd on my phone so I am wondering about that

meteyou, 2021-02-18:
i cannot reduce the update interval, because these are push updated. there is no polling. if you fill out the full bugfix form and upload all data i can reopen this ticket and double check it. with this minimal description, i cannot check anything...

meteyou, 2021-02-18:
this is the current dev version of mainsail and the current fluidd version of app.fluidd.xyz:

[image]

i can't see any difference between the two, unfortunately.

SilverWx, 2021-02-20:
Thank you a lot for checking that.
I admit I have not using fluidd a long time, I just remember I didn't get problem on my phone but I have to try again.
I'm not at home for now but I will check and get log trying using mainsail on my phone and post it here.

On my phone, at the beginning all work in fluid way at the beginning but after clicking move axis button multiple time it start lagging.

meteyou, 2021-02-20:
i will try to switch off individual panels to see what the real problem is. i know that the webcam + the tempchart are major performance eating features.

TravisWilder, 2021-06-11:
I can confirm high cpu with Safari and chrome on Mac
without webcam and tempchart it is better, but it is still to high: 40% for just the dashboard is too much

meteyou, 2021-06-11:
@TravisWilder we are working on this issue...

meteyou, 2021-07-16:
new beta release should fix the issue. pls test it

closed 2021-08-01 by meteyou