mainsail-crew/mainsail #207: [FR] webcam url to also have option to be img tag or an iframe tag

madsi1m, 2021-02-22:
**Is your feature request related to a problem? Please describe.**
I would like to embed camera(s) using iframe rather than using an img tag so i can have more control of the camera feed.

**Describe the solution you'd like**
I would like the option to specify either an iframe or img for the webcam url

**Describe alternatives you've considered**
Javascript hack to change img tag to an iframe. Probably better to do it right from the start.

**Additional context**
N/A

meteyou, 2021-02-22:
i am currently working on several variations of the webcam in general. however, i personally see iframe as a bad variant for a webcam. it can also be "exploited" very easily.

madsi1m, 2021-02-22:
I think an iframe is fine for a camera if the said camera has its own framework/eco system like say a video conferencing camera.
I also agree that iframes can be exploited very easily depending on setup of headers/security. Doesn't mean it shouldn't be an option.

meteyou, 2021-03-17:
currently i have no interest in iframes for such a purpose. if you think of another idea to integrate camera control, feel free to open a new feature request.

closed 2021-03-17 by meteyou