mainsail-crew/mainsail #211: Update Manager showing Invalid status and only klipper+moonraker

mylesagray, 2021-02-24:
**Describe the bug:**\
Updated to latest on moonraker, klipper and mainsail (running on mainsailos) - Enabled update_manager in config:

```ini
[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: True
config_path: ~/klipper_config

[authorization]
enabled: True
trusted_clients:
 10.0.0.0/8
 127.0.0.1
 169.254.0.0/16
 172.16.0.0/12
 192.168.0.0/16
 FE80::/10
 ::1/128

[update_manager]
enable_auto_refresh: true
repo: meteyou/mainsail
path: ~/mainsail
```

The mainsail UI shows the following:

[image]

All folders (`~/klipper`, `~/moonraker` and `~/mainsail`) all have valid git remotes and pulls work as expected.

**Expected behavior:**\
Moonraker discovers all packages and pulls latest versions from upstream repos.

**Screenshots:**\
Above

**Logfiles:**\
https://gist.github.com/mylesagray/b38b06ec8a2901ca1b96992979faee12

meteyou, 2021-02-27:
pls upload the full moonraker.log...

mylesagray, 2021-02-28:
@meteyou sorry about that, full moonraker and kliippy log included in Gist.

mylesagray, 2021-02-28:
Fixed it... the `moonraker.log` was showing that the repos weren't on `master` branch - I don't recall checking out a different branch, but anyway.

I did a `git checkout master` on each repo and a `git pull`.

`moonraker` was failing due to a missing dep (I was running on an older mainsailos release), so I used the commands in the mainsailos image build to install the deps:

https://github.com/raymondh2/MainsailOS/blob/256682d0088f96d60bcb27a672cd325405d0ddfb/src/modules/klipper_moonraker/filesystem/home/pi/install-custompios.sh#L87

```sh
KLIPPER_PYTHON_DIR="${HOME}/klippy-env"
KLIPPER_SOURCE_DIR="${HOME}/klipper"
MOONRAKER_PYTHON_DIR="${HOME}/moonraker-env"
MOONRAKER_SOURCE_DIR="${HOME}/moonraker"
CONFIG_PATH="/home/pi/klipper_config/moonraker.conf"
SYSTEMDDIR="/etc/systemd/system"
${KLIPPER_PYTHON_DIR}/bin/pip install -r ${KLIPPER_SOURCE_DIR}/scripts/klippy-requirements.txt
${MOONRAKER_PYTHON_DIR}/bin/pip install -r ${MOONRAKER_SOURCE_DIR}/scripts/moonraker-requirements.txt
```

After this it was all working as expected, closing.

closed 2021-02-28 by mylesagray