mainsail-crew/mainsail #218: [FR] Go to

Hochdruck3D, 2021-02-28:
**Is your feature request related to a problem? Please describe.**
Since I often want to drive to an exact coordinate and this is not so possible with the +5 buttons and about the Gcodes but very cumbersome I wish a text field in which you can enter the coordinate simply

**Describe the solution you'd like**
I wish a text field in which you can enter the coordinate simply

th33xitus, 2021-02-28:
> about the Gcodes but very cumbersome

So you find it less cumbersome to switch through at least 3 individual input fields you have to fill out (x, y, z and optionally f) instead of just typing the G1 command in a single line in the console input?

So doing it like this is cumbersome:
* Move cursor to console input
* input "G1 X123 Y456 F6000"
* click send

But like this isn't:
* Move cursor to x input field
* input x value
* move cursor to y input field
* input y value
* move cursor to f input field
* input f value
* click send

Hmmm... not quite sure. Or am i missing somehow the true benefit of such a feature?
The control buttons are also customizable now, you can define the steps those buttons do by yourself.

itCarl, 2021-03-17:
@Hochdruck3D 
Simply create a gcode Macro if u have to go to a specific coordinate very often ¯\_(ツ)_/¯

just like the following:

[gcode_macro X150_Y150_Z200]
   # make sure to use absolute positioning
   G90
   # Go to X=150 Y=150, would be center if bed is 300mm in x and y
   # and Go to Z = 200 
   G0 X150 Y150 Z200

ssendev, 2022-03-06:
Here is a macro that does this.

It remembers it's last speed so it's less to type on the cli.
Moves outside the allowed area are clamped to the stepper min/max position.
Moves occur above a configured min height except if z is passed then it moves directly to that point

```jinja
[gcode_macro TO]
# params: X Y Z S(mm/sec) F(mm/min)
# moves to the given X Y Z coordinate
# Stops at configured stepper min/max positions instead of ignoring the move
variable_speed: 6000 # default speed in mm/min, can be changed with S or F param
variable_default_position: False # e.g. "X10 Y10" False for no default
variable_min_z_height: 5 # when Z is not passed ensure moves occur above this height
variable_min_z_speed: 600 # z speed for rising to min_z_height
gcode:
    SAVE_GCODE_STATE NAME=my_move_to_state
    G90
    {% set min_x = printer.configfile.config["stepper_x"]["position_min"]|float %}
    {% set min_y = printer.configfile.config["stepper_y"]["position_min"]|float %}
    {% set min_z = printer.configfile.config["stepper_z"]["position_min"]|float %}
    {% set max_x = printer.configfile.config["stepper_x"]["position_max"]|float %}
    {% set max_y = printer.configfile.config["stepper_y"]["position_max"]|float %}
    {% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}
    {% set z_hop = False %}
    {% set args = "" %}

    {% if params.X %}
      {% set x = params.X|float %}
      {% set x = min_x if x < min_x else max_x if x > max_x else x  %}
      {% set args = args ~ " X" ~ x  %}
    {% endif %}
    {% if params.Y %}
      {% set y = params.Y|float %}
      {% set y = min_y if y < min_y else max_y if y > max_y else y  %}
      {% set args = args ~ " Y" ~ y  %}
    {% endif %}
    {% if params.Z %}
      {% set z = params.Z|float %}
      {% set z = min_z if z < min_z else max_z if z > max_z else z  %}
      {% set args = args ~ " Z" ~ z  %}
    {% else %}
      {% set tool_z = printer.toolhead.position.z %}
      # {% set min_z_height = printer.configfile.config["safe_z_home"]["z_hop"]|float %}
      {% if tool_z < min_z_height %}
        {% set z_hop = True %}
        # {% set min_z_speed = printer.configfile.config["safe_z_home"]["z_hop_speed"]|float %}
        G1 Z{min_z_height} F{min_z_speed}
      {% endif %}
    {% endif %}

    {% if params.S %}
      {% set f = params.S|float * 60 %}
      SET_GCODE_VARIABLE MACRO=TO VARIABLE=speed VALUE={f}
      {% set args = args ~ " F" ~ f  %}
    {% elif params.F %}
      {% set f = params.F|float %}
      SET_GCODE_VARIABLE MACRO=TO VARIABLE=speed VALUE={f}
      {% set args = args ~ " F" ~ f  %}
    {% else %}
      {% if args == "" and default_position %}
        {% set args = "F" ~ speed ~ " " ~ default_position %}
      {% else %}
        {% set args = args ~ " F" ~ speed  %}
      {% endif %}
    {% endif %}

    G1 {args}

    {% if z_hop %}
      G1 Z{tool_z} F{min_z_speed}
    {% endif %}
    RESTORE_GCODE_STATE NAME=my_move_to_state
```

closed 2022-04-24 by meteyou