mainsail-crew/mainsail #222: PrusaSlice not working Print Host upload

Foxbiker, 2021-03-06:
**Describe the bug:**\
After all updates is not possible send job from PrusaSlice to mainsailos. 

**Expected behavior:**\
[moonraker (2).log](https://github.com/meteyou/mainsail/files/6095075/moonraker.2.log)
[klippy.log](https://github.com/meteyou/mainsail/files/6095076/klippy.log)

**Screenshots:**\
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
- klippy.log
- moonraker.log

fabe83, 2021-03-06:
The behavior is due to a change in the moonraker.
In the moonraker.cfg the line [octoprint_compat] has to be added.
Further details are available here: https://github.com/Arksine/moonraker/blob/master/docs/user_changes.md

closed 2021-03-11 by meteyou

Foxbiker, 2021-03-11:
It help, thank you.