mainsail-crew/mainsail #224: Config Changes not persistent

DerInternets, 2021-03-11:
First of all: Thumbs up for Mainsail, i love it! 
I've just upgraded to the recent version of mainsail (1.4.0), so there's now options for temp presets, a movement cross, etc. - nice!

Unfortunately, Changes are not persistent - as soon as I refresh the page or reopen the browser, all changes are lost and I don't see a "save mainsail config" - button or something similar.

meteyou, 2021-03-11:
pls upload your moonraker.log. sounds like you use a old version of Moonraker.

DerInternets, 2021-03-11:
you're correct. updating moonraker fixed the issue. Unfortunately, It seems that i've created a minor new one: the update manager section in Machine Settings now shows moonraker version as ? and the status as invalid. Any hints?

DerInternets, 2021-03-11:
[moonraker (4).log](https://github.com/meteyou/mainsail/files/6125454/moonraker.4.log)

meteyou, 2021-03-11:
```
2021-03-11 20:12:23,262 [update_manager.py:initialize()] - Git Repo moonraker: Initialization failure
Traceback (most recent call last):
  File "/home/luke/moonraker/moonraker/plugins/update_manager.py", line 765, in initialize
    await self.fetch()
  File "/home/luke/moonraker/moonraker/plugins/update_manager.py", line 883, in fetch
    f"{self.git_cmd} fetch {self.git_remote}")
  File "/home/luke/moonraker/moonraker/plugins/update_manager.py", line 993, in _do_fetch_pull
    raise self.server.error(f"Git Command '{cmd}' failed")
utils.ServerError: Git Command 'git -C /home/luke/moonraker fetch origin' failed
2021-03-11 20:12:23,262 [update_manager.py:refresh()] - Error Refreshing git state
Traceback (most recent call last):
  File "/home/luke/moonraker/moonraker/plugins/update_manager.py", line 541, in refresh
    await self._update_repo_state()
  File "/home/luke/moonraker/moonraker/plugins/update_manager.py", line 548, in _update_repo_state
    await self.repo.initialize(need_fetch=need_fetch)
  File "/home/luke/moonraker/moonraker/plugins/update_manager.py", line 765, in initialize
    await self.fetch()
  File "/home/luke/moonraker/moonraker/plugins/update_manager.py", line 883, in fetch
    f"{self.git_cmd} fetch {self.git_remote}")
  File "/home/luke/moonraker/moonraker/plugins/update_manager.py", line 993, in _do_fetch_pull
    raise self.server.error(f"Git Command '{cmd}' failed")
utils.ServerError: Git Command 'git -C /home/luke/moonraker fetch origin' failed
```

moonraker cannot read the git status of the moonraker dir. pls post the output of:
```
cd ~/moonraker
git status
```

DerInternets, 2021-03-11:
`On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
`

meteyou, 2021-03-13:
you can try this: https://docs.mainsail.xyz/update/moonraker#update-moonraker-dependence

DerInternets, 2021-03-16:
No change, unfortunately.

meteyou, 2021-03-16:
Can you make a video?

DerInternets, 2021-03-17:
what do you wanna see? :)

meteyou, 2021-03-17:
- at first safe the output of http://printer.local/server/database/item?namespace=mainsail
- make a video of change something in settings > interface
- safe another output of http://printer.local/server/database/item?namespace=mainsail
- refresh mainsail
- safe another output of http://printer.local/server/database/item?namespace=mainsail

and then send the video and all 3 outputs of the moonraker db.

closed 2021-04-20 by meteyou