mainsail-crew/mainsail #227: Remote Mode Printer missing from Printers page until browser refresh

mcarter14777, 2021-03-13:
I have been blindly poking my way around remote mode and my Voron V0 rarely shows up on the list and I have to refresh. Then when it does appear, the printer name is not displayed, only the address and port, and the printer status says unknown. My Switchwire shows up on the list correctly every time with printer name and status. I have changed the hostnames of each printer so they are not duplicates. I know the docs for remote mode are not done yet but I feel like I should share my experience.

Thanks for your awesome work!
Let me know if you need me to provide any logs!

meteyou, 2021-03-13:
are both printers on the current Moonraker version? the printer name in settings > interface will be displayed in the printers page. the hostname is only to connect...

mcarter14777, 2021-03-13:
Both printers are on the latest versions. On the printers page I have seen it display the printer card using the printer name and also the status with temperatures under, but only a couple times. I believe everything is working fine connection wise as I do get the status in the drop-down menu. The issues are most likely with the printers page.

meteyou, 2021-08-01:
should be fixed

closed 2021-08-01 by meteyou