mainsail-crew/mainsail #228: [FR] Select Date/ time format in g-code files menu

stenc55, 2021-03-13:
**Is your feature request related to a problem? Please describe.**
Not a problem, annoyance.

**Describe the solution you'd like**
Somehow to switch to standard dd.mm.yy 24h format in g-code files window. Or, if it already exists now, clear instructions how to do it. Maybe pull format from raspberry?

[image]

meteyou, 2021-03-15:
this should be read out of your system, but i have to check it. i use the javascript function "toLocalString" -> https://github.com/meteyou/mainsail/blob/master/src/pages/Files.vue#L513

stenc55, 2021-03-15:
In Windows I have set CET timezone, it shows as "Monday, 15. March 2011
In Raspi, it shows "Mon 15 Mar 15:00 GMT 2021". I have to set timezone..
If that helps?

closed 2021-03-26 by meteyou