mainsail-crew/mainsail #229: Reverse Y

lilianzzz, 2021-03-14:
Please also add reverse Y. On the plate Anycubic mega S (trigorilla) Forward is backward and backward is forward on Y. Everything is fine with the print, it's a matter of manual control

meteyou, 2021-03-15:
if your stepper goes in the wrong direction, invert your dir_pin in your printer.cfg or change the cables of your stepper. your printer also prints mirror-inverted if you dont change it.

zellneralex, 2021-03-15:
Do me an favor type 
G90
G0 X0 Y0
and Tell us where the head moves. To my understanding that should be the corner front left of your build plate

lilianzzz, 2021-03-15:
[image]
[image]

Everything is fine when printing.  Parking at the near left corner.  this is a feature of the board, one of the drivers is inverted, the screenshot shows the standard config and mine is with TMC2208

meteyou, 2021-03-15:
with G0 X0 Y0, the noozle is on the front left?

lilianzzz, 2021-03-15:
> with G0 X0 Y0, the noozle is on the front left?

Yep

meteyou, 2021-03-15:
and wenn you press the upper arrow, the toolhead goes to the back of printarea?

lilianzzz, 2021-03-16:
This is prusa kinematics. X0 Y0 is in the front left corner. When i press the upper arrow, table moves to the far left corner (X0 Y210)

meteyou, 2021-03-16:
Yes that's right. You move your toolhead back. Your relativ point is the toolhead and not the bed!

lilianzzz, 2021-03-17:
A matter of convenience. it is more convenient for someone to navigate relative to the table, and not the toolhead. And this option is available in fluidd and octoprint for a reason... For the same reason, the shutdown of the motors is already present as a separate button. Just for convenience. And for convenience, I switched from fluidd to mainsail, at low resolutions your interface is much more concise and user-friendly. especially from the latest version with alternative controls.

meteyou, 2021-03-17:
i didn't want to include y invert on purpose, so as not to tempt the user to bypass a wrong klipper config with the gui correction...
and it is always concerned with the toolhead, with every movement, and not with the individual components... but if other users speak up, i'll have to implement it anyway....

meteyou, 2021-03-26:
invert option for all 3 axis, will be in the next release.

closed 2021-03-26 by meteyou

lilianzzz, 2021-03-27:
> invert option for all 3 axis, will be in the next release.

Thank you very much for your work