mainsail-crew/mainsail #230: [FR] Adjust Z-Offset when not printing

DruckiMcDruck, 2021-03-15:
Klipper is able to accept Z-offset settings before (or after) a print. Currently mainsail only shows the z-offset box while printing. I'd like to adjust the z-offset via the box before starting a print, as i sometimes know that i need a z offset before i hit "print". 

Also being able to change the available offset options would be awesome.

meteyou, 2021-03-15:
there will be an option to enable this panel the whole time in the next release. 

I personally see the problem somewhere else if you know it beforehand anyway, but since it is now very often requested, I will add it.

DruckiMcDruck, 2021-03-15:
It's mostly just me being lazy. I could either change my endstop position physically on the printer or inside printer.cfg, but instead i sometimes just change the z-offset before starting a print. 

Thank you for adding it :)

Stephan3, 2021-03-15:
I use zoffset too outside of printing state.

commit fafd1055 (2021-03-15): feature: add option to display ZOffsetPanel in Standby (fix #230)
closed 2021-03-15 by meteyou