mainsail-crew/mainsail #236: Multiple instances - overwrite config

1walkerit, 2021-03-20:
I'm using multiple instances of mainsail. Ater update to new version I am not able to save config for each printer. The confgi is overwriter by printers. I think it is because the config moved to moonraker database - This release focused on migrating from .mainsail.json to the new moonraker database. 
Is there any way how to solve it?
Thank you for help

meteyou, 2021-03-20:
can you explain your issue?

1walkerit, 2021-03-20:
> can you explain your issue?

I am not able to have independent settings for each printer. If I rename printer 1, it also renames printers 2 and 3 and theri settings (camera settings, preheat ...). If I made changes to printer 2, it also change the same way at printer 1 and 3 ..and so on.
this was before update:
[image]

and just now:
[image]
all three printers the same name and settings

meteyou, 2021-03-23:
do you use KIAUH with multi instances of moonraker on the same PI?

1walkerit, 2021-03-24:
> do you use KIAUH with multi instances of moonraker on the same PI?

yes, I've installed using KIAUH script and using multiple moonrake on different ports (7125-7128) on the same Pi

meteyou, 2021-03-24:
ahh ok. this was a bug in KIAUH. it forget to change the moonraker_db for each instance -> https://github.com/th33xitus/kiauh/commit/6de8ce32787859ad5d0d4297f865c258093e8ef1

maybe @th33xitus can help you to fix this issue.

th33xitus, 2021-03-24:
Unless you don't want to delete your existing moonraker.conf files you have to edit them manually and add those different database_path to the configs. Otherwise, if you don't have customized those moonraker.confs any further you can also just delete them and re-install moonraker with KIAUH again. That's your choice now.

If you want to edit the files manually, open up each moonraker.conf and put in the following line under config_path:
`database_path: ~/.moonraker_database_$INSTANCE` where `$INSTANCE` is the number of that printer instance.

So in your case you have 4 instances set up, that means:
moonraker.conf in printer_1 needs:
`database_path: ~/.moonraker_database_1`

moonraker.conf in printer_2 needs:
`database_path: ~/.moonraker_database_2`

moonraker.conf in printer_3 needs:
`database_path: ~/.moonraker_database_3`

moonraker.conf in printer_4 needs:
`database_path: ~/.moonraker_database_4`

After that, restart all moonraker instances. You can do that from within KIAUH. Just start KIAUH and type "restart moonraker" in the "Perform action" input where you usually put in a number.

1walkerit, 2021-03-24:
GREAT, thank you for your help. Working fine (y)

closed 2021-03-24 by 1walkerit